Azure-Samples/Custom-vision-service-iot-edge-raspberry-pi #6: Build Solution fails

saradotramli, 2018-07-30:
> Please provide us with the following information:
> ---------------------------------------------------------------

### This issue is for a: (mark with an `x`)
```
- [x ] bug report -> please search issues before submitting
- [ ] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
> Upon invoking "Build IoT Edge Solution" command from VS Code

### Any log messages given by the failure
>
Sending build context to Docker daemon  12.08MB
Step 1/12 : FROM mohaseeb/raspberrypi3-python-opencv:latest
Get https://registry-1.docker.io/v2/mohaseeb/raspberrypi3-python-opencv/manifests/latest: unauthorized: incorrect username or password

### Expected/desired behavior
> Build should succeed

### OS and Version?
> Windows 7, 8 or 10. Linux (which distribution). macOS (Yosemite? El Capitan? Sierra?)
Windows 10, VSCode 1.25.1, Docker for Windows Community Edition Version 18.06.0-ce-win72 (19098)
Channel: stable

### Versions
>

### Mention any other details that might be useful

> ---------------------------------------------------------------
> Thanks! We'll be in touch soon.

saradotramli, 2018-08-01:
Hi, any updates on this?

ebertrams, 2018-08-07:
Closing since the docker logout command should have fixed it. Let me know if this is not the case.

closed 2018-08-07 by ebertrams