Azure-Samples/Custom-vision-service-iot-edge-raspberry-pi #8: Build failed: Unable to locate package libboost-python1.55.0

qbertz, 2018-08-08:
> Please provide us with the following information:
> ---------------------------------------------------------------

### This issue is for a: (mark with an `x`)
```
- [X] bug report -> please search issues before submitting
- [ ] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
> Build IoT Edge Solution in Visual Studio Code (v1.25.1) on "deployment.template.json"

### Any log messages given by the failure
>Step 4/12 : RUN apt-get install -y         libboost-python1.55.0
 ---> Running in c6...
Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package libboost-python1.55.0
E: Couldn't find any package by regex 'libboost-python1.55.0'
The command '/bin/sh -c apt-get install -y         libboost-python1.55.0' returned a non-zero code: 100

### Expected/desired behavior
> Successful build.

### OS and Version?
> Windows 10

### Versions
>latest

### Mention any other details that might be useful

> ---------------------------------------------------------------
> Thanks! We'll be in touch soon.

ebertrams, 2018-08-08:
Thanks for reporting, we're investigating

ebertrams, 2018-08-10:
Hi @qbertz, Could you please give some more details on which docker image you re trying to build exactly? Also could you please check that you have the latest code synced and check your internet connectivity?

I can't repro your issue and I dont see why it would not be able to get the libboost package.

Here is what I get when building the arm32 docker image of the camera-capture image, at step 4:

Step 4/13 : RUN apt-get install -y libboost-python1.55.0
 ---> Running in 3655cff96cae
Reading package lists...
Building dependency tree...
Reading state information...
Suggested packages:
  python3
The following NEW packages will be installed:
  libboost-python1.55.0
0 upgraded, 1 newly installed, 0 to remove and 11 not upgraded.
Need to get 128 kB of archives.
After this operation, 511 kB of additional disk space will be used.
Get:1 http://deb.debian.org/debian/ jessie/main libboost-python1.55.0 armhf 1.55.0+dfsg-3 [128 kB]
debconf: delaying package configuration, since apt-utils is not installed
Fetched 128 kB in 3s (35.4 kB/s)
Selecting previously unselected package libboost-python1.55.0.
(Reading database ... 18339 files and directories currently installed.)
Preparing to unpack .../libboost-python1.55.0_1.55.0+dfsg-3_armhf.deb ...
Unpacking libboost-python1.55.0 (1.55.0+dfsg-3) ...
Setting up libboost-python1.55.0 (1.55.0+dfsg-3) ...
Processing triggers for libc-bin (2.19-18+deb8u10) ...
Removing intermediate container 3655cff96cae
 ---> 9aa8a8382218
Step 5/13 : COPY /build/arm32v7-requirements.txt ./

Thanks

qbertz, 2018-08-10:
Hi @ebertrams,

removing previous images and pulling a fresh Alpine:latest base made this issue pass away. 

Many thanks.

closed 2018-08-10 by qbertz