Azure-Samples/Custom-vision-service-iot-edge-raspberry-pi #11: Can't Build TensorFlow in ImageClassifierService

danieljsmarda, 2018-08-14:
> Please provide us with the following information:
> ---------------------------------------------------------------

### This issue is for a: (mark with an `x`)
```
- [X] bug report -> please search issues before submitting
- [ ] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
> Right click deployment.template.json --> click "Build IoT Edge Solution"

### Any log messages given by the failure
>
[image]

### Expected/desired behavior
> Successful build

### OS and Version?
> Windows 10 Enterprise, Version 1709, OS Build 16299.551

### Versions
> VS Code v1.24.0, Docker version 18.03.1-ce, build 9ee9f40
### Mention any other details that might be useful
1. The SenseHatDisplay and CameraCapture modules have both built successfully; I am only having problems with the ImageClassifierService module.
2. I've tried to build on two different networks (one was a mobile hotspot, the other was a corporate network with a proxy that I haven't had problems with in the past) and received the same error when trying with both networks.

Thank you in advance for your help!

> ---------------------------------------------------------------
> Thanks! We'll be in touch soon.

ebertrams, 2018-08-14:
Yes as of yesterday the TensorFlow team deprecated their TF wheel installer. I believe that it now has been replaced by a pip installer on Raspberry Pi and we're investigating this option.

ebertrams, 2018-08-15:
I just updated the master branch with a fix to install TensorFlow on Raspberry Pi per their new guidelines: https://www.tensorflow.org/install/install_raspbian.

closed 2018-08-15 by ebertrams

danieljsmarda, 2018-08-16:
Confirmed that it's built and working, thank you!