Azure-Samples/Custom-vision-service-iot-edge-raspberry-pi #13: cv2.error: OpenCV(3.4.1) /opencv/opencv-3.4.1/modules/imgproc/src/resize.cpp:4044: error: (-215) ssize.width > 0 && ssize.height > 0 in function resize

Softwaremaker, 2018-08-19:
> Please provide us with the following information:
> ---------------------------------------------------------------

### This issue is for a: (mark with an `x`)
```
- [X] bug report -> please search issues before submitting
- [ ] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
> Only difference in the project as opposed to what was demonstrated in the YouTube video is that I am using an on-board camera on the Raspi3

### Any log messages given by the failure
>OpenCV(3.4.1) Error: Assertion failed (ssize.width > 0 && ssize.height > 0) in resize, file /opencv/opencv-3.4.1/modules/imgproc/src/resize.cpp, line 4044
Original frame size: 0x0
Frame rate (FPS): 0
Frame number: 1
Time to capture (+ straighten up) a frame: 0 ms
Traceback (most recent call last):
  File "./main.py", line 125, in <module>
    main(VIDEO_PATH, IMAGE_PROCESSING_ENDPOINT, IMAGE_PROCESSING_PARAMS, SHOW_VIDEO, VERBOSE, LOOP_VIDEO, CONVERT_TO_GRAY, RESIZE_WIDTH, RESIZE_HEIGHT, ANNOTATE)
  File "./main.py", line 94, in main
    cameraCapture.start()
  File "/CameraCapture.py", line 162, in start
    preprocessedFrame = cv2.resize(frame, (self.resizeWidth, self.resizeHeight))
cv2.error: OpenCV(3.4.1) /opencv/opencv-3.4.1/modules/imgproc/src/resize.cpp:4044: error: (-215) ssize.width > 0 && ssize.height > 0 in function resize

### Expected/desired behavior
It seems that the OpenCV2 module is not detecting an input camera stream. Not sure if I am using an on-board camera as opposed to a USB Camera is the problem.

In any case, I am using the test video: AppleAndBanana.mp4. Here are my createOptions:
Camera Capture Azure IoT Edge Module. Press Ctrl-C to exit.
Initialising the camera capture with the following parameters:
   - Video path: /home/pi/share/AppleAndBanana.mp4
   - Image processing endpoint: http://127.0.0.1:80/
   - Image processing params: ""
   - Show video: False
   - Loop video: True
   - Convert to gray: False
   - Resize width: 256
   - Resize height: 256
   - Annotate: False
   - Send processing results to hub: True

### OS and Version?
Raspberry Pi3 with VS Code
For the camera capture module, I am pulling the image from ebertraacr.azurecr.io/cameracapture:0.2.5-arm32v7

### Versions
>

### Mention any other details that might be useful

> ---------------------------------------------------------------
> Thanks! We'll be in touch soon.

ebertrams, 2018-08-20:
Hi @Softwaremaker .

Yes the issue is with the camera that you re using. The camera capture module does not detect your camera (Original frame size: 0x0). Only USB cameras are supported in this sample. Could you please to a USB camera?

The /home/pi/share/AppleAndBanana.mp4 video path also will likely not unless you ve given it is access to the camera-capture container.

sridhar21111976, 2019-02-27:
Hi, Can you please confirm if the camera redirection works for Windows 10 PC running desktop docker with the camera Capture module...? I am getting the below error

cv2.error: OpenCV(4.0.0) /io/opencv/modules/imgproc/src/resize.cpp:3784: error: (-215:Assertion failed) !ssize.empty() in function 'resize'

ebertrams, 2019-02-28:
USB pass through is on Docker for Windows's backlog but is currently not supported. Alternative is to stream a video feed to your docker VM via a port binding.