Azure-Samples/Custom-vision-service-iot-edge-raspberry-pi #18: Camera-Capture module crash on Windows Docker

wrende, 2018-09-18:
> Please provide us with the following information:
> ---------------------------------------------------------------

### This issue is for a: (mark with an `x`)
```
- [X] bug report -> please search issues before submitting
- [ ] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
> Deploy amd64 solution on a Windows 10 simulated IoT Edge Device.

### Any log messages given by the failure
> The camera-capture module will crash on runtime with no logging. All the other modules run with logging available.
```cmd
PS C:\WINDOWS\system32> iotedge logs camera-capture
PS C:\WINDOWS\system32>
```
```cmd
PS C:\WINDOWS\system32> iotedge list
NAME                      STATUS           DESCRIPTION            CONFIG
edgeHub                   running          Up an hour             mcr.microsoft.com/azureiotedge-hub:1.0
image-classifier-service  running          Up an hour             fcscontainers.azurecr.io/imageclassifierservice:0.2.4-amd64
camera-capture            stopped          Stopped 4 minutes ago  fcscontainers.azurecr.io/cameracapture:0.2.7-test-amd64
edgeAgent                 running          Up an hour             mcr.microsoft.com/azureiotedge-agent:1.0
```

### Expected/desired behavior
>A running camera-capture container with logging on a windows 10 device with docker installed (followed the [instructions on Windows](https://docs.microsoft.com/en-us/azure/iot-edge/how-to-install-iot-edge-windows-with-linux)).

### OS and Version?
> Windows 10

### Versions
> Docker Version 17.12.0-ce-win47 (15139)
> VS Code version 1.27
> PowerShell version 5.1.16299.611

### Mention any other details that might be useful
Can't provide detailed info because there is no logging available.

> ---------------------------------------------------------------
> Thanks! We'll be in touch soon.

ebertrams, 2018-09-18:
Hi @wrende , thanks for reporting this bug.

I am assuming that you use LInux containers on your Windows 10 machine. Could you please share the the logs of the edgeAgent, edgeHub and cameraCapture module( docker logs ...)  so that we can investigate?

Thanks

m-chau, 2018-09-19:
Hi @wrende, do you mind also provide the output of `docker ps -a`? Thank you.

wrende, 2018-09-19:
Thanks for the quick reply's.

Found the issue, the docker file test-amd64 of the camera-capture modules has issue's with Windows 10 Docker engine. Probably a wrong entrypoint.

I fixed it by editing the amd64 deployment file to use the camera-capture amd64 dockerfile (not the test-amd64).
Now it runs smooth without issues on a Windows 10 Device with docker.

Tip: Dont forget to add "ADD /test/ ." to your docker file if you want to use the .mp4

baatch, 2018-09-19:
@wrende 

Did you manage to get it working with a live webcam feed or are you using the video?
What value did you set camera-caputre modules  "createOptions" on the deployment.json file to?

I changed to amd64 dockerfile but still can't get it to work in Windows 10 PC.

wrende, 2018-09-20:
@baatch 

Got it working only with the video. 
Also please correct me if I'm wrong but I don't think you can let the camera-module work on a Windows 10 device + USB cam. If it is possible please update the camera-module createoptions doc.

Here is my createoption to let it work with a video:

```json
"{\"Env\":[\"VIDEO_PATH=./AppleAndBanana.mp4\",\"IMAGE_PROCESSING_ENDPOINT=http://image-classifier-service:80/image\", \"VERBOSE=TRUE\", \"RESIZE_WIDTH=256\", \"RESIZE_HEIGHT=256\", \"SHOW_VIDEO=TRUE\"], \"HostConfig\":{\"PortBindings\":{\"5012/tcp\":[{\"HostPort\":\"5012\"}]}}}"
```

baatch, 2018-09-20:
Ok then I understand. Nowhere does it say that Live webcam feed will not work for the Windows 10 PC option.

ebertrams, 2018-09-20:
Correct, this is a limitation of Docker for Windows. You cant use read streams from a USB device. Reading a stream from a USB device will only work on a Linux host that has direct access to the USB camera (e.g. not the MobyLinuxVM set up by Docker for Windows) with Linux containers.

Also correct that the test-amd64 does not run the python script by default by just open a bash prompt. This is mostly useful to debug your own custom version of this container.

closed 2018-10-05 by ebertrams