Azure-Samples/Custom-vision-service-iot-edge-raspberry-pi #19: IoTEdge HSM Failures [RPI3]

wrende, 2018-09-25:
> Please provide us with the following information:
> ---------------------------------------------------------------

### This issue is for a: (mark with an `x`)
```
- [ X ] bug report -> please search issues before submitting
- [ ] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
> Follow the installation guide --> [instructions to install Azure IoT Edge](https://docs.microsoft.com/en-us/azure/iot-edge/how-to-install-iot-edge-linux-arm)

### Any log messages given by the failure

```shell
Sep 25 16:20:38 raspberrypi iotedged[4154]: 2018-09-25T14:20:38Z [INFO] - Checking edge runtime status
Sep 25 16:20:38 raspberrypi iotedged[4154]: 2018-09-25T14:20:38Z [INFO] - Edge runtime is running.
Sep 25 16:20:42 raspberrypi iotedged[4154]: 2018-09-25T14:20:42Z [ERR!] - Internal server error: Sign failed
Sep 25 16:20:42 raspberrypi iotedged[4154]:         caused by: A error occurred retrieving a key from the key store.
Sep 25 16:20:42 raspberrypi iotedged[4154]:         caused by: HSM failure
Sep 25 16:20:42 raspberrypi iotedged[4154]:         caused by: HSM API failure occurred: 364
Sep 25 16:20:42 raspberrypi iotedged[4154]: 2018-09-25T14:20:42Z [INFO] - [work] - - - [2018-09-25 14:20:42.378014093 UTC] "POST /modules/%24edgeAgent/genid/636734724559850094/encrypt?api-version=2018-06-28 HTTP/1.1" …3 "-" "-" pid(4361)
Sep 25 16:20:47 raspberrypi iotedged[4154]: 2018-09-25T14:20:47Z [INFO] - [mgmt] - - - [2018-09-25 14:20:47.455864768 UTC] "GET /modules?api-version=2018-06-28 HTTP/1.1" 200 OK 962 "-" "-" pid(4361)
Sep 25 16:20:47 raspberrypi iotedged[4154]: 2018-09-25T14:20:47Z [ERR!] - Internal server error: Sign failed
Sep 25 16:20:47 raspberrypi iotedged[4154]:         caused by: A error occurred retrieving a key from the key store.
Sep 25 16:20:47 raspberrypi iotedged[4154]:         caused by: HSM failure
Sep 25 16:20:47 raspberrypi iotedged[4154]:         caused by: HSM API failure occurred: 364
Sep 25 16:20:47 raspberrypi iotedged[4154]: 2018-09-25T14:20:47Z [INFO] - [work] - - - [2018-09-25 14:20:47.459371101 UTC] "POST /modules/%24edgeAgent/genid/636734724559850094/encrypt?api-version=2018-06-28 HTTP/1.1" …3 "-" "-" pid(4361)
Sep 25 16:20:47 raspberrypi iotedged[4154]: t open file for reading /var/lib/iotedge/hsm/enc_keys/edgelet-masterWt5mT2xpO72EPKlt2Tt0Sq4uJCrMvfl2rzzKRB3pnyo_.enc.key. Errno 2 'No such file or directory'
Sep 25 16:20:47 raspberrypi iotedged[4154]: 2018-09-25T14:20:25Z [ERR!] (/project/hsm-sys/azure-iot-hsm-c/src/edge_hsm_client_store.c:load_encryption_key_from_file:824) Could not read key from file. Key size 0
Sep 25 16:20:47 raspberrypi iotedged[4154]: 2018-09-25T14:20:25Z [ERR!] (/project/hsm-sys/azure-iot-hsm-c/src/edge_hsm_client_store.c:edge_hsm_client_open_key:2145) HSM store could not load encryption key edgelet-master
Sep 25 16:20:47 raspberrypi iotedged[4154]: 2018-09-25T14:20:25Z [ERR!] (/project/hsm-sys/azure-iot-hsm-c/src/edge_hsm_client_crypto.c:encrypt_data:363) Could not get encryption key by name 'edgelet-master'
Sep 25 16:20:47 raspberrypi iotedged[4154]: 2018-09-25T14:20:25Z [ERR!] (/project/hsm-sys/azure-iot-hsm-c/src/hsm_utils.c:read_file_into_buffer_impl:166) Could not open file for reading /var/lib/iotedge/hsm/enc_keys/e… file or directory'
Sep 25 16:20:47 raspberrypi iotedged[4154]: 2018-09-25T14:20:25Z [ERR!] (/project/hsm-sys/azure-iot-hsm-c/src/edge_hsm_client_store.c:load_encryption_key_from_file:824) Could not read key from file. Key size 0
Sep 25 16:20:47 raspberrypi iotedged[4154]: 2018-09-25T14:20:25Z [ERR!] (/project/hsm-sys/azure-iot-hsm-c/src/edge_hsm_client_store.c:edge_hsm_client_open_key:2145) HSM store could not load encryption key edgelet-master
Sep 25 16:20:47 raspberrypi iotedged[4154]: 2018-09-25T14:20:25Z [ERR!] (/project/hsm-sys/azure-iot-hsm-c/src/edge_hsm_client_crypto.c:encrypt_data:363) Could not get encryption key by name 'edgelet-master'
Sep 25 16:20:47 raspberrypi iotedged[4154]: 2018-09-25T14:20:30Z [ERR!] (/project/hsm-sys/azure-iot-hsm-c/src/hsm_utils.c:read_file_into_buffer_impl:166) Could not open file for reading /var/lib/iotedge/hsm/enc_keys/e… file or directory'
Sep 25 16:20:47 raspberrypi iotedged[4154]: 2018-09-25T14:20:30Z [ERR!] (/project/hsm-sys/azure-iot-hsm-c/src/edge_hsm_client_store.c:load_encryption_key_from_file:824) Could not read key from file. Key size 0
Sep 25 16:20:47 raspberrypi iotedged[4154]: 2018-09-25T14:20:30Z [ERR!] (/project/hsm-sys/azure-iot-hsm-c/src/edge_hsm_client_store.c:edge_hsm_client_open_key:2145) HSM store could not load encryption key edgelet-master
Sep 25 16:20:47 raspberrypi iotedged[4154]: 2018-09-25T14:20:30Z [ERR!] (/project/hsm-sys/azure-iot-hsm-c/src/edge_hsm_client_crypto.c:encrypt_data:363) Could not get encryption key by name 'edgelet-master'
Sep 25 16:20:47 raspberrypi iotedged[4154]: 2018-09-25T14:20:42Z [ERR!] (/project/hsm-sys/azure-iot-hsm-c/src/hsm_utils.c:read_file_into_buffer_impl:166) Could not open file for reading /var/lib/iotedge/hsm/enc_keys/e… file or directory'
Sep 25 16:20:47 raspberrypi iotedged[4154]: 2018-09-25T14:20:42Z [ERR!] (/project/hsm-sys/azure-iot-hsm-c/src/edge_hsm_client_store.c:load_encryption_key_from_file:824) Could not read key from file. Key size 0
Sep 25 16:20:47 raspberrypi iotedged[4154]: 2018-09-25T14:20:42Z [ERR!] (/project/hsm-sys/azure-iot-hsm-c/src/edge_hsm_client_store.c:edge_hsm_client_open_key:2145) HSM store could not load encryption key edgelet-master
Sep 25 16:20:47 raspberrypi iotedged[4154]: 2018-09-25T14:20:42Z [ERR!] (/project/hsm-sys/azure-iot-hsm-c/src/edge_hsm_client_crypto.c:encrypt_data:363) Could not get encryption key by name 'edgelet-master'
Sep 25 16:20:47 raspberrypi iotedged[4154]: 2018-09-25T14:20:47Z [ERR!] (/project/hsm-sys/azure-iot-hsm-c/src/hsm_utils.c:read_file_into_buffer_impl:166) Could not open file for reading /var/lib/iotedge/hsm/enc_keys/e… file or directory'
Sep 25 16:20:47 raspberrypi iotedged[4154]: 2018-09-25T14:20:47Z [ERR!] (/project/hsm-sys/azure-iot-hsm-c/src/edge_hsm_client_store.c:load_encryption_key_from_file:824) Could not read key from file. Key size 0
Sep 25 16:20:47 raspberrypi iotedged[4154]: 2018-09-25T14:20:47Z [ERR!] (/project/hsm-sys/azure-iot-hsm-c/src/edge_hsm_client_store.c:edge_hsm_client_open_key:2145) HSM store could not load encryption key edgelet-master
Sep 25 16:20:47 raspberrypi iotedged[4154]: 2018-09-25T14:20:47Z [ERR!] (/project/hsm-sys/azure-iot-hsm-c/src/edge_hsm_client_crypto.c:encrypt_data:363) Could not get encryption key by name 'edgelet-master'
Sep 25 16:20:47 raspberrypi iotedged[4154]: 2018-09-25T14:20:47Z [ERR!] (/project/hsm-sys/azure-iot-hsm-c/src/hsm_utils.c:read_file_into_buffer_impl:166) Could not open file for reading /var/lib/iotedge/hsm/enc_keys/e… error: Sign failed
```

### Expected/desired behavior
> Working IoTEdge daemon with no encryption errors, for some reason I'm missing some files

### OS and Version?
> Raspberry Pi 3
```shell
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
```

### Mention any other details that might be useful
Could be due to an installation error that the certificates are missing (timing issues maybe)
Is there a command that will refresh all the IoT Edge certificate files ?

> ---------------------------------------------------------------
> Thanks! We'll be in touch soon.

ebertrams, 2018-09-25:
Hi @wrende , thanks for reporting this bug.

For us to investigate, could you please specify which kind of device you re using? In particular, does it have a specific HSM/TPM?

Thanks!

wrende, 2018-09-26:
Not that I know, it is a clean RPI3 with rasbian. However I have solved the problem by formatting and starting over again. This issue probably was caused due to an NTP issue (network blocked ntp packages) at the install of the IoT Edge Security Daemon. This caused a creation of expired certificates. Is it possible to renew the certificates created in the /var/lib/iotedge/hsm ?

ebertrams, 2018-10-05:
I dont know of a way to recreate the certificates. Closing this issue for now since rebooting fixed it.

closed 2018-10-05 by ebertrams

AdamMachera, 2019-03-06:
The way I did it:
apt-get remove --purge iotedge
rm -rf  /var/lib/iotedge
apt-get install iotedge

update configuration e.g.:
vim /etc/iotedge/config.yaml

start iotedge
/etc/init.d/iotedge start