Azure-Samples/Custom-vision-service-iot-edge-raspberry-pi #23: Error when building

smoms, 2018-10-15:
Hi,

when i try to build and push the solution i get error:

Sending build context to Docker daemon  12.09MB
Step 1/12 : FROM resin/raspberrypi3-python:2.7
Get https://registry-1.docker.io/v2/resin/raspberrypi3-python/manifests/2.7: unauthorized: incorrect username or password

Please help.

smoms, 2018-10-15:
So i had to run `docker login` first...

ebertrams, 2018-10-15:
Yes, this is expected.

1. For docker login, this is what is meant by: https://github.com/Azure-Samples/Custom-vision-service-iot-edge-raspberry-pi#to-deploy-the-solution-on-a-raspberry-pi-3, "make sure that your docker engine has access to it"
2. Building the camera-capture module can take several hours depending on your machine. Open-cv, numpy and pillow are particularly long to build.

closed 2018-10-15 by ebertrams

smoms, 2018-10-17:
ok thanks. Now after hours it ends up with another error:

The push refers to repository [azurecontainerxxxxx.azurecr.io/cameracapture]
203c50c19ed3: Preparing
18d3493f95df: Preparing
5c412b3f3b9f: Preparing
31115809f800: Preparing
05e19133d7a8: Preparing
2562252fcab1: Preparing
9bff5e3cfb03: Preparing
558c9f90dcd6: Preparing
5771be8f06de: Preparing
3e34a36df651: Preparing
84578535dd7d: Preparing
ad8faa962060: Preparing
28e56680b5a3: Preparing
e9a72e5741fa: Preparing
aaeda15e2adb: Preparing
7fd0cf42f554: Preparing
101ec3719318: Preparing
8132d72e0029: Preparing
ac01d333e084: Preparing
db3e5dca97da: Preparing
dc8bcc6ab3b9: Waiting
7ce7082a1de0: Waiting
a3e1630cb500: Waiting
210e1176810d: Waiting
52f7e619f26a: Waiting
8d5bafbf9be5: Waiting
c69d7c20a132: Waiting
e28e07db7a43: Waiting
655595dfb2a2: Waiting
e631cfda4e31: Waiting
c26bbb0d565f: Waiting
3afc57049c19: Waiting
700eafb3a88b: Waiting
unauthorized: authentication required

In my .env file i have the variables:

CONTAINER_REGISTRY_ADDRESS="azurecontainerxxxx.azurecr.io"
CONTAINER_REGISTRY_USERNAME="azurecontainerxxxx"
CONTAINER_REGISTRY_PASSWORD="xxxxxx/yyyyyyMxxxxxxS"

what is wrong?
thanks

smoms, 2018-10-17:
Ok thanks. Now after hours of building i get error at last step:

The push refers to repository [azurecontainerxxxx.azurecr.io/cameracapture]
203c50c19ed3: Preparing
18d3493f95df: Preparing
5c412b3f3b9f: Preparing
31115809f800: Preparing
05e19133d7a8: Preparing
2562252fcab1: Preparing
9bff5e3cfb03: Preparing
558c9f90dcd6: Preparing
5771be8f06de: Preparing
3e34a36df651: Preparing
84578535dd7d: Preparing
ad8faa962060: Preparing
28e56680b5a3: Preparing
2562252fcab1: Pushed
aaeda15e2adb: Pushed
7fd0cf42f554: Retrying in 1 second
101ec3719318: Pushed
8132d72e0029: Pushed
ac01d333e084: Pushed
db3e5dca97da: Pushed
dc8bcc6ab3b9: Pushed
7ce7082a1de0: Pushed
a3e1630cb500: Pushed
210e1176810d: Pushed
52f7e619f26a: Pushed
8d5bafbf9be5: Pushed
c69d7c20a132: Pushed
e28e07db7a43: Pushed
655595dfb2a2: Pushed
e631cfda4e31: Pushed
c26bbb0d565f: Pushed
3afc57049c19: Pushed
700eafb3a88b: Pushed
net/http: TLS handshake timeout

can you please help?
thanks

ebertrams, 2018-10-17:
It looks like a networking issue. Can you verify your internet connection and try again (docker should have cached all the layers so build time should just be a few seconds now if you have not changed anything)?

smoms, 2018-10-17:
thanks. i made a _docker login <acr server>_ and solved!