Azure-Samples/Custom-vision-service-iot-edge-raspberry-pi #24: Camera-capture module never runs on Ubuntu

zambottivinicius, 2018-10-22:
- [ x] bug report
- [ ] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)

### Minimal steps to reproduce
> I'm trying to run this demo on a linux PC and I've created an container registry and uploaded both containers (Image and camera) and also a Linux VM to use as a device. For some reason when I deploy the IoT Edge solution the camera module never starts, even the container does not start or throw any error. And when I start to monitor D2C messages, nothing happens. 

### Any log messages given by the failure
[image]

[image]

[image]

### OS and Version?
> Ubuntu 16.04

ebertrams, 2018-10-22:
Can you try rebooting the device? And could you please provide the logs of the camera-capture module (docker logs camera-capture)?

zambottivinicius, 2018-10-22:
I already restarted my device several times using sudo systemctl restart iotedge. Regarding the logs,they're empty.

[image]

zambottivinicius, 2018-10-22:
That's my deployment.json 
`{
  "modulesContent": {
    "$edgeAgent": {
      "properties.desired": {
        "schemaVersion": "1.0",
        "runtime": {
          "type": "docker",
          "settings": {
            "minDockerVersion": "v1.25",
            "loggingOptions": "",
            "registryCredentials": {
              "registryName": {
                "username": "xxxx",
                "password": "xxxx",
                "address": "xxx"
              }
            }
          }
        },
        "systemModules": {
          "edgeAgent": {
            "type": "docker",
            "settings": {
              "image": "mcr.microsoft.com/azureiotedge-agent:1.0",
              "createOptions": ""
            }
          },
          "edgeHub": {
            "type": "docker",
            "status": "running",
            "restartPolicy": "always",
            "settings": {
              "image": "mcr.microsoft.com/azureiotedge-hub:1.0",
              "createOptions": ""
            }
          }
        },
        "modules": {
          "camera-capture": {
            "version": "1.0",
            "type": "docker",
            "status": "running",
            "restartPolicy": "always",
            "settings": {
              "image": "logiotedge.azurecr.io/cameracapture:0.2.7-test-amd64",
              "createOptions": "{\"Env\":[\"VIDEO_PATH=./AppleAndBanana.mp4\",\"IMAGE_PROCESSING_ENDPOINT=http://image-classifier-service:80/image\", \"VERBOSE=TRUE\", \"RESIZE_WIDTH=256\", \"RESIZE_HEIGHT=256\", \"SHOW_VIDEO=TRUE\"], \"HostConfig\":{\"PortBindings\":{\"5012/tcp\":[{\"HostPort\":\"5012\"}]}}}"
            }
          },
          "image-classifier-service": {
            "version": "1.0",
            "type": "docker",
            "status": "running",
            "restartPolicy": "always",
            "settings": {
              "image": "logiotedge.azurecr.io/imageclassifierservice:0.2.4-amd64",
              "createOptions": ""
            }
          }
        }
      }
    },
    "$edgeHub": {
      "properties.desired": {
        "schemaVersion": "1.0",
        "routes": {
          "CameraCaptureToIoTHub": "FROM /messages/modules/camera-capture/outputs/output1 INTO $upstream"
        },
        "storeAndForwardConfiguration": {
          "timeToLiveSecs": 7200
        }
      }
    }
  }
}
`

ebertrams, 2018-10-29:
It could be because the test-amd64 images does not run the python script but just a open a bash prompt by default. Try:

- using
 the amd64 image directly and add "ADD /test/ ." to the docker file to include your test video file.

OR

- Changing the entry point of the test-amd64.Dockerfile to ENTRYPOINT [ "python", "-u", "./main.py" ] instead of ENTRYPOINT [ "bash" ]

zambottivinicius, 2018-10-31:
@ebertrams The first solution worked! thank you so much for the help.

closed 2018-10-31 by ebertrams