Azure-Samples/Custom-vision-service-iot-edge-raspberry-pi #27: Connection between the container modules fails due to a certificate error

MaWa19, 2018-12-04:
### This issue is for a: (mark with an `x`)
```
- [x] bug report -> please search issues before submitting
- [ ] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
Cloned the sample, updated env, built and pushed the deployment.template.json, created the deployment for my edge device. Modules are being deployed to the device but it seems like the Camera Capture Module can't establish a connection to the Image Classifier Service.

### Any log messages given by the failure
Camera Capture Module:
> Python 2.7.15 (default, Jun 22 2018, 18:43:23)
[GCC 4.9.2]

> Camera Capture Azure IoT Edge Module. Press Ctrl-C to exit.
ImageServer::Started.
Error: Time:Tue Dec  4 17:41:14 2018 File:/usr/sdk/src/c/c-utility/adapters/tlsio_openssl.c Func:send_handshake_bytes Line:661 error:14090086:SSL routines:SSL3_GET_SERVER_CERTIFICATE:certificate verify failed
Error: Time:Tue Dec  4 17:41:14 2018 File:/usr/sdk/src/c/c-utility/adapters/tlsio_openssl.c Func:send_handshake_bytes Line:661 error:14090086:SSL routines:SSL3_GET_SERVER_CERTIFICATE:certificate verify failed
Error: Time:Tue Dec  4 17:41:16 2018 File:/usr/sdk/src/c/c-utility/adapters/tlsio_openssl.c Func:send_handshake_bytes Line:661 error:14090086:SSL routines:SSL3_GET_SERVER_CERTIFICATE:certificate verify failed
Error: Time:Tue Dec  4 17:41:20 2018 File:/usr/sdk/src/c/c-utility/adapters/tlsio_openssl.c Func:send_handshake_bytes Line:661 error:14090086:SSL routines:SSL3_GET_SERVER_CERTIFICATE:certificate verify failed
Error: Time:Tue Dec  4 17:41:28 2018 File:/usr/sdk/src/c/c-utility/adapters/tlsio_openssl.c Func:send_handshake_bytes Line:661 error:14090086:SSL routines:SSL3_GET_SERVER_CERTIFICATE:certificate verify failed
Error: Time:Tue Dec  4 17:41:44 2018 File:/usr/sdk/src/c/c-utility/adapters/tlsio_openssl.c Func:send_handshake_bytes Line:661 error:14090086:SSL routines:SSL3_GET_SERVER_CERTIFICATE:certificate verify failed

### Expected/desired behavior
Running custom vision service.

### OS and Version?
Raspbian stretch 9.6

### Mention any other details that might be useful
It seems like the SenseHatDisplay module is not pushed to the Azure Container Registry and therefore can't be found by the Edge Device. However, I don't have such a display on my pi anyway so I don't really care about that module.

MaWa19, 2018-12-05:
In the meantime I set up another Pi with a clean installation of the newest Raspbian and Edge-Runtime and got it running on this device. So the error above might be related to incompatabilites with previous versions of the IotEdge-Runtime.

closed 2018-12-05 by MaWa19

ebertrams, 2018-12-05:
@MaWa19  got it, thanks for the follow up!