Azure-Samples/Custom-vision-service-iot-edge-raspberry-pi #30: Annotate attribute in CameraCapture not working for object detection

ShehataHaroun, 2019-01-22:
### This issue is for a: (mark with an `x`)
```
- [x] bug report -> please search issues before submitting
- [ ] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```
Annotate function does not work for object detection, video stream will no be displayed if you set annotate to TRUE.
### Minimal steps to reproduce
Set annotate Attribute to True in CameraCapture.py
Add "ANNOTATE":{"value": "True"} to CameraCapture module env ind deployment.template.json

### Any log messages given by the failure
No logs but inside the Code there is a TODO:

### Expected/desired behavior
Should Display Object Detection Boundry Box on Image Server

### OS and Version?
Rasbian 9.6

### Versions

### Mention any other details that might be useful
I had to change the dockerfile of the CameraCapture module for it to work
[image]

ShehataHaroun, 2019-01-31:
Here Is a Quick fix for Object Detection I did.
In CameraCapture.py
Extract the right Properties from the response json.

`self.response = [(x["boundingBox"]["width"], x["boundingBox"]["top"],x["boundingBox"]["height"],x["boundingBox"]["left"], x["probability"]) for x in <RESPONSE STRING VARIABLE HERE>["predictions"]]`

Adapted _annotate Method.
`    def __annotate(self, frame):
        if self.response != []:
            for rectangle in self.response:
                if rectangle[4] >= 0.65:
                    top_calc = rectangle[1] * self.height
                    left_calc = rectangle[3] * self.width
                    width_calc = rectangle[0] * self.width
                    height_calc = rectangle[2]* self.height 
                    #print(rectangle[1],rectangle[3],rectangle[0],rectangle[2])
                    #print(top_calc,left_calc,width_calc,height_calc)
                    cv2.rectangle(frame, (int(left_calc),int(top_calc)), (int(left_calc+ width_calc),int(top_calc+height_calc)), (70,70,70),2)
                    cv2.putText(frame, "Probability: "+ str(int(rectangle[4]*100)) +"%",(int(left_calc),int(top_calc)),cv2.FONT_HERSHEY_PLAIN,1.0,(204,0,51), 1,4)`

This should display rectangles insider the frame.

closed 2019-01-31 by ShehataHaroun

flaz09092018, 2019-03-04:
Hello, could you give me some more details about the steps needed to make this work? I am experimenting with the x64 Version. Thanks in advande

ShehataHaroun, 2019-03-05:
Hello,
I did this with the x32 Version and made some changes in the HTTP request, so I am not a 100% sure it will work for you. But in general I took these steps to display the boundry box received from the response of a object detection model (will not work with image classification):
1. Set the Annotate flag in the deploymet.template.json to false
2. Define this attribute inside the def__init__() method
`self.response = []`
3. Add these two line after the HTTP post in the _sendFrameForProcessing Method
`json_data= response.json()`
`self.response = [(x["boundingBox"]["width"], x["boundingBox"]["top"],x["boundingBox"]["height"],x["boundingBox"]["left"], x["probability"]) for x in json_data["predictions"]]`
4. Replace the Annotate method with
`    def __annotate(self, frame):
        if self.response != []:
            for rectangle in self.response:
                if rectangle[4] >= 0.65:
                    top_calc = rectangle[1] * self.height
                    left_calc = rectangle[3] * self.width
                    width_calc = rectangle[0] * self.width
                    height_calc = rectangle[2]* self.height 
                    #print(rectangle[1],rectangle[3],rectangle[0],rectangle[2])
                    #print(top_calc,left_calc,width_calc,height_calc)
                    cv2.rectangle(frame, (int(left_calc),int(top_calc)), (int(left_calc+ width_calc),int(top_calc+height_calc)), (70,70,70),2)
                    cv2.putText(frame, "Probability: "+ str(int(rectangle[4]*100)) +"%",(int(left_calc),int(top_calc)),cv2.FONT_HERSHEY_PLAIN,1.0,(204,0,51), 1,4)`
5. Update the method calls inside the start method in the Display frames section
`self.__annotate(frame)`

I hope these steps work for you :)

Kind Regards
Haroun