Azure-Samples/Custom-vision-service-iot-edge-raspberry-pi #36: Executor failed to create kernel error when running on raspberry

FundamentX, 2019-03-08:
I deployed this solution over 30 times on my Raspberry Pi 3 over several months now. Just recently when i build and deployed the solution again the "ImageClassifierService" won't process any images anymore. Tested it with postman and it returns "Error processing image".

If i look at the logs of the module it say's:

/usr/lib/python3.5/importlib/_bootstrap.py:222: RuntimeWarning: compiletime version 3.4 of module 'tensorflow.python.framework.fast_tensor_util' does not match runtime version 3.5
  return f(*args, **kwds)

/usr/lib/python3.5/importlib/_bootstrap.py:222: RuntimeWarning: builtins.type size changed, may indicate binary incompatibility. Expected 432, got 412
  return f(*args, **kwds)

WARNING:tensorflow:From /app/predict.py:28: FastGFile.__init__ (from tensorflow.python.platform.gfile) is deprecated and will be removed in a future version.
Instructions for updating:
Use tf.gfile.GFile.

Loading model...Success!
Loading labels...2 found. Success!
 * Serving Flask app "app" (lazy loading)
 * Environment: production
   WARNING: Do not use the development server in a production environment.
   Use a production WSGI server instead.
 * Debug mode: off
 * Running on http://0.0.0.0:80/ (Press CTRL+C to quit)

2019-03-08 01:52:14.047760: E tensorflow/core/common_runtime/executor.cc:624] Executor failed to create kernel. Not found: No registered '_FusedConv2D' OpKernel for CPU devices compatible with node {{node conv1/conv1}}
        .  Registered:  <no registered kernels>

Can anybody help me with this? I downloaded and rebuild the entire solution a couple of times with no success.

FundamentX, 2019-03-08:
If i specify tensorflow==1.12.0 in the ImageClassifierService module, it works.

ebertrams, 2019-03-08:
Thanks! I added your fix to the repo.

closed 2019-03-08 by ebertrams