Azure-Samples/Custom-vision-service-iot-edge-raspberry-pi #37: RPI3: Executor failed to create kernel _FusedConv2D

njbuch, 2019-03-10:
> Please provide us with the following information:
> ---------------------------------------------------------------

### This issue is for a: (mark with an `x`)
```
- [x ] bug report -> please search issues before submitting
- [ ] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
- download the complete project
- build and push
- deploy and check logs

### Any log messages given by the failure
>
Caused by op 'conv1/conv1', defined at:
  File "/usr/lib/python3.5/threading.py", line 882, in _bootstrap
    self._bootstrap_inner()
  File "/usr/lib/python3.5/threading.py", line 914, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.5/threading.py", line 862, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/lib/python3.5/socketserver.py", line 625, in process_request_thread
    self.finish_request(request, client_address)
  File "/usr/lib/python3.5/socketserver.py", line 354, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/lib/python3.5/socketserver.py", line 681, in __init__
    self.handle()
  File "/usr/local/lib/python3.5/dist-packages/werkzeug/serving.py", line 293, in handle
    rv = BaseHTTPRequestHandler.handle(self)
  File "/usr/lib/python3.5/http/server.py", line 422, in handle
    self.handle_one_request()
  File "/usr/local/lib/python3.5/dist-packages/werkzeug/serving.py", line 328, in handle_one_request
    return self.run_wsgi()
  File "/usr/local/lib/python3.5/dist-packages/werkzeug/serving.py", line 270, in run_wsgi
    execute(self.server.app)
  File "/usr/local/lib/python3.5/dist-packages/werkzeug/serving.py", line 258, in execute
    application_iter = app(environ, start_response)
  File "/usr/local/lib/python3.5/dist-packages/flask/app.py", line 2309, in __call__
    return self.wsgi_app(environ, start_response)
  File "/usr/local/lib/python3.5/dist-packages/flask/app.py", line 2292, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.5/dist-packages/flask/app.py", line 1813, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.5/dist-packages/flask/app.py", line 1799, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "app.py", line 41, in predict_image_handler
    results = predict_image(img)
  File "/app/predict.py", line 56, in predict_image
    tf.import_graph_def(graph_def, name='')
  File "/usr/local/lib/python3.5/dist-packages/tensorflow/python/util/deprecation.py", line 507, in new_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.5/dist-packages/tensorflow/python/framework/importer.py", line 442, in import_graph_def
    _ProcessNewOps(graph)
  File "/usr/local/lib/python3.5/dist-packages/tensorflow/python/framework/importer.py", line 235, in _ProcessNewOps
    for new_op in graph._add_new_tf_operations(compute_devices=False):  # pylint: disable=protected-access
  File "/usr/local/lib/python3.5/dist-packages/tensorflow/python/framework/ops.py", line 3433, in _add_new_tf_operations
    for c_op in c_api_util.new_tf_operations(self)
  File "/usr/local/lib/python3.5/dist-packages/tensorflow/python/framework/ops.py", line 3433, in <listcomp>
    for c_op in c_api_util.new_tf_operations(self)
  File "/usr/local/lib/python3.5/dist-packages/tensorflow/python/framework/ops.py", line 3325, in _create_op_from_tf_operation
    ret = Operation(c_op, self)
  File "/usr/local/lib/python3.5/dist-packages/tensorflow/python/framework/ops.py", line 1801, in __init__
    self._traceback = tf_stack.extract_stack()

NotFoundError (see above for traceback): No registered '_FusedConv2D' OpKernel for CPU devices compatible with node node conv1/conv1 (defined at /app/predict.py:56) 
	.  Registered:  <no registered kernels>

	 [[node conv1/conv1 (defined at /app/predict.py:56) ]]
2019-03-10 14:44:50.408362: E tensorflow/core/common_runtime/executor.cc:624] Executor failed to create kernel. Not found: No registered '_FusedConv2D' OpKernel for CPU devices compatible with node {{node conv1/conv1}}
	.  Registered:  <no registered kernels>

### Expected/desired behavior
>The image classifier should be able to classify images... ;)

### OS and Version?
> Building and pushing from Win 10. Running edge device is Rpi 3+

### Versions
>

### Mention any other details that might be useful
The camera capture fails because it cannot create the socket to the image-classifier service.

This might be useful: https://github.com/tensorflow/tensorflow/issues/24732

> ---------------------------------------------------------------
> Thanks! We'll be in touch soon.

nanowireUK, 2019-03-11:
Saw same issue as described. Fix was changing line in requirements.txt and rebuilding
`tensorflow==1.12.0`

ebertrams, 2019-03-11:
Yes this commit shoudl have fixed it: https://github.com/Azure-Samples/Custom-vision-service-iot-edge-raspberry-pi/commit/cf80d80bfd439ef5f30f17ca5bba50382a67ed4c#diff-49f375eebd2ba1e403c3a586f22c22fb 

Please sync your repro and rebuild your solution. Let me know if you still hit this issue.

closed 2019-03-11 by ebertrams