Azure-Samples/Custom-vision-service-iot-edge-raspberry-pi #48: E: You don't have enough free space in /var/cache/apt/archives/.

arashaga, 2019-05-10:
When I do build and push to IoT hub from VS Code. I get the following error:

E: You don't have enough free space in /var/cache/apt/archives/.

I did df-h on my Raspberry device and got the following: 

Filesystem      Size  Used Avail Use% Mounted on
/dev/root        30G  5.1G   23G  19% /
devtmpfs        460M     0  460M   0% /dev
tmpfs           464M   13M  452M   3% /dev/shm
tmpfs           464M   36M  429M   8% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           464M     0  464M   0% /sys/fs/cgroup
/dev/mmcblk0p1   44M   22M   22M  51% /boot
tmpfs            93M  4.0K   93M   1% /run/user/1000

so it does not seems like I do not have space. 
 the version of Raspbian is :
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
 Also I am trying to push from VS code installed on Windows 10.

Any help will be appreciate it.

ebertrams, 2019-07-31:
I have not seen this error before. Have you tried a "docker system prune -a"?

ebertrams, 2019-12-10:
Closing since no reply

closed 2019-12-10 by ebertrams