Azure-Samples/Custom-vision-service-iot-edge-raspberry-pi #50: image-classifier-module fails on AMD64 image - module 'tensorflow' has no attribute 'GraphDef'

nanowireUK, 2019-06-11:
> Please provide us with the following information:
> ---------------------------------------------------------------

### This issue is for a: (mark with an `x`)
```
- [x] bug report -> please search issues before submitting
- [ ] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
> Clone repo
> Change deployment.template.json to AMD64 version
> Build and push IoT solution
> Let containers run and see image-classifier-service fails
> `iotedge logs image-classifier-service`

### Any log messages given by the failure
Traceback (most recent call last):
  File "app.py", line 15, in <module>
    from predict import initialize, predict_image, predict_url
  File "/app/predict.py", line 22, in <module>
    graph_def = tf.GraphDef()
AttributeError: module 'tensorflow' has no attribute 'GraphDef'

### Expected/desired behavior
> It works with included Dockerfile

### OS and Version?
> Ubuntu 18.04 and Windows

### Versions
>

### Mention any other details that might be useful
Can be fixed by changing the Dockerfile to use a different base image and then install fixed tensorflow 1.12.0 and then including the various apt install for python dependencies.
#FROM tensorflow/tensorflow:latest-py3
FROM ubuntu:xenial
> ---------------------------------------------------------------
> Thanks! We'll be in touch soon.

nielsams, 2019-10-16:
@nanowireUK 
I had the same issue and was able to do a quick fix by adding 'tensorflow==1.12.0' to the end of the amd64-requirements.txt file and rebuilding. 

Awaiting a proper fix of course, but maybe this will help you forward for now

ebertrams, 2019-12-10:
Fixed with https://github.com/Azure-Samples/Custom-vision-service-iot-edge-raspberry-pi/pull/62

closed 2019-12-10 by ebertrams