Azure-Samples/Custom-vision-service-iot-edge-raspberry-pi #60: image-classifier fails with illigal instruction on AMD64 'illegal instruction' when importing tensorflow as tf

initmahesh, 2019-12-05:
> Please provide us with the following information:
> ---------------------------------------------------------------

### This issue is for a: (mark with an `x`)
```
- [ ] bug report -> please search issues before submitting
- [ ] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
> Clone this sample
Update the .env file with the values for your container registry and make sure that your docker engine has access to it
3.Build the entire solution by opening the control palette (Ctrl+Shift+P), select Build and push IoT Edge Solution (this can take a while...especially to build numpy and pillow...) and select the deployment.test-amd64.template.json manifest file
4.Deploy the solution to your device by right-clicking on the config/deployment.json file, select Create Deployment for Single device and choose your targeted device
verify module image-classifier-service  are not running.
create module image-classifier-service failed in docker.

### Any log messages given by the failure
>
upsquared@UP2:~$ sudo docker ps
CONTAINER ID        IMAGE                                                COMMAND                   CREATED             STATUS              PORTS                         NAMES
2153c5a76718        mycapreg.azurecr.io/cameracapture:0.2.8-test-amd64   "python3 -u ./main.py"    6 minutes ago       Up 2 seconds        0.0.0.0:5012->5012/tcp        camera-capture
e8a3815db9c2        mcr.microsoft.com/azureiotedge-hub:1.0               "/bin/sh -c 'echo \"$…"   25 minutes ago      Up 4 seconds        443/tcp, 5671/tcp, 8883/tcp   edgeHub
0cf73814128b        mcr.microsoft.com/azureiotedge-agent:1.0             "/bin/sh -c 'echo \"$…"   7 days ago          Up 12 seconds                                     edgeAgent
upsquared@UP2:~$ sudo docker ps -a
CONTAINER ID        IMAGE                                                    COMMAND                   CREATED             STATUS                       PORTS                         NAMES
2153c5a76718        mycapreg.azurecr.io/cameracapture:0.2.8-test-amd64       "python3 -u ./main.py"    6 minutes ago       Up 9 seconds                 0.0.0.0:5012->5012/tcp        camera-capture
e8a3815db9c2        mcr.microsoft.com/azureiotedge-hub:1.0                   "/bin/sh -c 'echo \"$…"   25 minutes ago      Up 10 seconds                443/tcp, 5671/tcp, 8883/tcp   edgeHub
e7fae1986de1        mycapreg.azurecr.io/imageclassifierservice:0.2.5-amd64   "/bin/sh -c 'python …"    25 minutes ago      Exited (132) 4 minutes ago                                 image-classifier-service
0cf73814128b        mcr.microsoft.com/azureiotedge-agent:1.0                 "/bin/sh -c 'echo \"$…"   7 days ago          Up 19 seconds                                              edgeAgent
upsquared@UP2:~$ sudo docker logs -f image-classifier-service --tail 200
Illegal instruction (core dumped)
Illegal instruction (core dumped)
Illegal instruction (core dumped)
Illegal instruction (core dumped)
Illegal instruction (core dumped)
Illegal instruction (core dumped)
Illegal instruction (core dumped)
Illegal instruction (core dumped)
upsquared@UP2:~$ sudo systemctl stop iotedge
 

upsquared@UP2:~$ sudo docker run -it --rm mycapreg.azurecr.io/imageclassifierservice:0.2.5-amd64
Illegal instruction (core dumped)
upsquared@UP2:~$ sudo docker run -it --rm mycapreg.azurecr.io/imageclassifierservice:0.2.5-amd64 bash

________                               _______________                
___  __/__________________________________  ____/__  /________      __
__  /  _  _ \_  __ \_  ___/  __ \_  ___/_  /_   __  /_  __ \_ | /| / /
_  /   /  __/  / / /(__  )/ /_/ /  /   _  __/   _  / / /_/ /_ |/ |/ / 
/_/    \___//_/ /_//____/ \____//_/    /_/      /_/  \____/____/|__/

WARNING: You are running this container as root, which can cause new files in
mounted volumes to be created as the root user on your host machine.

To avoid this, run the container by specifying your user's userid:

$ docker run -u $(id -u):$(id -g) args...

root@4a0538a0028b:/app# python3 
Python 3.6.8 (default, Aug 20 2019, 17:12:48) 
[GCC 8.3.0] on linux

>>> import tensorflow as tf
Illegal instruction (core dumped)

### Expected/desired behavior
>

### OS and Version?
> Linux ubuntu 18.04
### Versions
>

### Mention any other details that might be useful

> ---------------------------------------------------------------
> Thanks! We'll be in touch soon.

initmahesh, 2019-12-05:
The issue seems to be that there is no AVX support as per here (https://stackoverflow.com/questions/49094597/illegal-instruction-core-dumped-after-running-import-tensorflow) that TF need beyond build 1.6 , ifwe go to build 1.5 we can get pass this issu ebut then operation tensorflow.compat.v1 is not supported as that got added in 1.14 onwards ….

ebertrams, 2019-12-10:
Fixed with https://github.com/Azure-Samples/Custom-vision-service-iot-edge-raspberry-pi/pull/62

closed 2019-12-10 by ebertrams