Azure-Samples/Custom-vision-service-iot-edge-raspberry-pi #63: String in deployment template

PeterHedley94, 2020-02-21:
> Please provide us with the following information:
> ---------------------------------------------------------------

### This issue is for a: (mark with an `x`)
```
- [x] bug report -> please search issues before submitting
- [ ] feature request
- [ ] documentation issue or request
- [x] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
> launch template on Visual Studio

### Any log messages given by the failure
>
```There are errors in deployment json file: data.modulesContent.$edgeAgent['properties.desired'].modules['camera-capture'].settings.createOptions should be string```

### Expected/desired behavior
> To upload correctly

### OS and Version?
Linux 19.0 and Visual Studio Code (Jan 2020 release)

### Versions
>

### Mention any other details that might be useful
I think you need to make the details like this...
```"{\"HostConfig\":{\"PortBindings\":{\"5012/tcp\":[{\"HostPort\":\"5012\"}]},\"Binds\":[\"/dev/video0:/dev/video0\"],\"Devices\":[{\"PathOnHost\":\"/dev/video0\",\"PathInContainer\":\"/dev/video0\",\"CgroupPermissions\":\"mrw\"}]}}"```
Or something that works and looks nice (very hard to see what is going on in that string). Also I think the file needs to be in a folder named config.
> ---------------------------------------------------------------
> Thanks! We'll be in touch soon.

ebertrams, 2020-02-21:
Hey @PeterHedley94,
Did you use the deployment.template.json file or the deployment.json one? You should start from the deployment.template.json file and generate the deployment.json file from there (using right click > generate manifest) and then use the deployment.json file to make your deployment (right-click > create deployment for single device)

PeterHedley94, 2020-02-22:
Thank you, new to iot edge and visual studio extensions. Closing now as is not a bug.

closed 2020-02-22 by PeterHedley94