Azure-Samples/Custom-vision-service-iot-edge-raspberry-pi #64: Deploy to RPi 4?

mtyeager, 2020-04-11:
> Please provide us with the following information:
> ---------------------------------------------------------------

### This issue is for a: (mark with an `x`)
```
- [ ] bug report -> please search issues before submitting
- [X] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
> Deployment to Raspberry Pi 4

### Any log messages given by the failure
>

### Expected/desired behavior
> Expect to follow the process as described to deploy a custom vision solution to an RPi 4. Are there known issues with deployment to a RPi 4?

### OS and Version?
> Windows 7, 8 or 10. Linux (which distribution). macOS (Yosemite? El Capitan? Sierra?)
Raspian 4.19
### Versions
>

### Mention any other details that might be useful

> ---------------------------------------------------------------
This is just an inquiry as to any known issues. Is there a plan to update this demo to use RPi4?:
> Thanks! We'll be in touch soon.

ebertrams, 2020-04-13:
Thanks for your question @mtyeager. Have you tried using the existing steps for Raspberry Pi 3 on a Raspberry Pi 4 device? I expect the same code to work but please let us know if it is not the case and please explain where it fails. Thank you!

mtyeager, 2020-04-18:
Hi Emmanuel,

I have deployed the modules to a RPi 4, but none run. They all exit and indicate “backed off”.

Am continuing to work on it.

Thanks,

Michael

mtyeager, 2020-04-18:
To be clear, the edge agent and hub are both running, it is the camera-capture, sense-hat and image-classifier that try to start but backoff.

mtyeager, 2020-04-21:
Yes that fixed it. THANKS!

closed 2020-04-21 by mtyeager