Azure-Samples/Custom-vision-service-iot-edge-raspberry-pi #65: Modules sensehat-display and image-classifier-service fail immediately

rgres, 2020-04-19:
> Please provide us with the following information:
> ---------------------------------------------------------------

### This issue is for a: (mark with an `x`)
```
- [x] bug report -> please search issues before submitting
- [ ] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
>Set up raspberry pi like in the discriptions (using Raspbian buster desktop), build and push IoT edge solution (VS code on Windows 10), deploy to raspberry pi, see modules "sensehat-display" and "image-classifier-service" fail (terminal "sudo iotedge list"), see logs of the two modules (terminal "sudo iotedge logs <moduleName>" 

### Any log messages given by the failure
> From Module image classifier service:

Traceback (most recent call last):
  File "app.py", line 10, in <module>
    from PIL import Image
  File "/usr/local/lib/python3.7/site-packages/PIL/Image.py", line 93, in <module>
    from . import _imaging as core
ImportError: libxcb.so.1: cannot open shared object file: No such file or directory

From Module sensehat-display:

Traceback (most recent call last):
  File "./main.py", line 14, in <module>
    import DisplayManager
  File "/DisplayManager.py", line 1, in <module>
    import sense_hat
  File "/usr/local/lib/python3.5/dist-packages/sense_hat/__init__.py", line 2, in <module>
    from .sense_hat import SenseHat, SenseHat as AstroPi
  File "/usr/local/lib/python3.5/dist-packages/sense_hat/sense_hat.py", line 14, in <module>
    from PIL import Image  # pillow
  File "/usr/local/lib/python3.5/dist-packages/PIL/Image.py", line 93, in <module>
    from . import _imaging as core
ImportError: libxcb.so.1: cannot open shared object file: No such file or directory

### Expected/desired behavior
>no log errors :D

### OS and Version?
> Windows 10, Linux (raspbian buster)

### Versions
>

### Mention any other details that might be useful
Tried this multiple times with fresh installs of the pi, on pi3 and pi4
> ---------------------------------------------------------------
> Thanks! We'll be in touch soon.

netsnake182, 2020-04-21:
Just found a fix for this. You need to add  `libgl1-mesa-glx` on line 5 for ..\modules\ImageClassifierService\arm32v7.Dockerfile and add `RUN apt update && apt install -y libgl1-mesa-glx` on line 9 for ..\modules\SenseHatDisplay\arm32v7.Dockerfile

[image]

build your images again, push to the device and happy days :-)

rgres, 2020-04-26:
That worked well, thank you

closed 2020-04-26 by rgres

ebertrams, 2020-04-28:
Merged PR 68 to master to fix this issue