Azure-Samples/Custom-vision-service-iot-edge-raspberry-pi #69: Asynchronous iot-device sdk

PeterHedley94, 2020-04-29:
> Please provide us with the following information:
> ---------------------------------------------------------------

### This issue is for a: (mark with an `x`)
```
- [ ] bug report -> please search issues before submitting
- [x] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
> 

### Any log messages given by the failure
>

### Expected/desired behavior
> Thank you for the great repo! The new SDK `azure-iot-device` not `azure-iot-device-client` supports and suggests using async requests. It would be great to get the repo updated for the new release. (I am struggling to get it to work).

### OS and Version?
> Windows 7, 8 or 10. Linux (which distribution). macOS (Yosemite? El Capitan? Sierra?)

### Versions
>

### Mention any other details that might be useful

> ---------------------------------------------------------------
> Thanks! We'll be in touch soon.

ebertrams, 2020-04-29:
Thanks for the suggestion Peter. It is true that this repo would benefit from switching to the new Azure SDK. I dont have enough time for now to make this switch but will keep it in mind and in any cases welcome contributions!

PeterHedley94, 2020-05-27:
@ebertrams been trying to find time to do this myself. Got it working a while ago now just cleaning it up. Will close this issue and do a PR when it is less hacky.

closed 2020-05-27 by PeterHedley94